hybridgroup/cylon-i2c #28: Any advice on edison + LSM9DS0

fooqri, 2015-02-13:
Thanks for the module, I have used it successfully with a Yun with MPU6050, but having a problem getting the gyro example to work.

My setup is an Intel Edison, LSM9DS0 (https://www.sparkfun.com/products/13033) and the cylon-intel-iot as the adapter.

i2c seems to have correct addresses on the edison for the gyro and accel/mag
# i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- 1d -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- 6b -- -- -- -- 
70: -- -- -- -- -- -- -- --           

I modified the gyro example as so

"use strict";

var Cylon = require("cylon");

Cylon
  .robot()
  .connection("edison", { adaptor: "intel-iot"})
  .device("gyro", { driver: "lsm9ds0g" })

  .on("ready", function(bot) {
    setInterval(function() {
      bot.gyro.getGyro(function(err, data) {
        console.log(data);
      });
    }, 1000);
  });

Cylon.start();

But the output is not correct, and doesn't change in relation to the movement of the device. 

I, [2015-02-12T23:58:31.435Z]  INFO -- : Initializing connections.
I, [2015-02-12T23:58:31.450Z]  INFO -- : Initializing devices.
I, [2015-02-12T23:58:31.725Z]  INFO -- : Starting connections.
I, [2015-02-12T23:58:31.730Z]  INFO -- : Starting devices.
I, [2015-02-12T23:58:31.741Z]  INFO -- : Working.
{ x: 8224, y: 8224, z: -515 }
{ x: 8224, y: 8224, z: -515 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }
{ x: 8224, y: 8224, z: 8297 }

The same output for similar changes to demo for getAccel. 

I appreciate any advice on approaches I might take to identify the issue and/or get this to work? I have limited experience with i2c on linux, so may be a basic mistake.

Thanks much!

deadprogram, 2015-02-13:
Hi, @fooqri how do you have the LSM9DS0 wired to the Edison?

Keep in mind that the SparkFun version of the LSM9DS0 is a 3.3V device, so you should make sure that you are not feeding it 5V.

fooqri, 2015-02-13:
Thanks for the quick response @deadprogram. 

Sparkfun has a special line of Edison ready component blocks that utilize the Hirose DF40 connector on the Edison, and can be chained together via the connector. There is an example of a stack of blocks and explanation of the blocks here: 

https://learn.sparkfun.com/tutorials/general-guide-to-sparkfun-blocks-for-intel-edison.

The schematic for the LSM9DS0 block is here: https://cdn.sparkfun.com/datasheets/Dev/Edison/9dof_block.pdf

The LSM9DS0 block is designed to pull from one of the 3.3V connectors on the Hirose DF40 connector.

My setup is a stack that includes an Edison, LSM9DS0 block, and power block. I have tried two different LSM9DS0 blocks, and several different power blocks and the result is the same. I believe Sparkfun has mad this nearly impossible to mess up on the hardware side, but I could be wrong. 

I am not sure that the cylon-intel-iot adaper is designed to work with the sparkfun board (it mentions edison's arduino dev board) but I am not sure what other adapter to use with cylon-i2c to connect to the edison. It would be great to get tis to work as Sparkfun has designed a really easy way to proto with these blocks.

Thanks.

deadprogram, 2015-02-15:
Hi, @fooqri yes, the Sparkfun blocks are quite easy to connect physically. However, the software support for them is a little different than the other options for connecting the i2c devices. I think @zankich has one of the blocks you have. I'm pretty sure we would just need a way to specify which i2c interface to use from cylon-intel-iot, to make this work. We'll check it out for sure.

edgarsilva, 2015-03-06:
Any updates on how to specify the I2C interface? @zankich do you have the block? have you had any chance to check this?

daniel-lcc, 2015-06-23:
Hi, regarding the above, I've also connected my edison with the sparkfun lsm9ds0 block and tried running the driver for lsm9ds0xm, but it gave me errors as follow:

This is my code, taken from http://cylonjs.com/documentation/drivers/lsm9ds0xm/ but using the edison intel iot adaptor:

"use strict";
var Cylon = require('cylon');
Cylon.robot({
	connections: {
		edison: { adaptor: 'intel-iot' }
	},

	devices: {
		accel: { driver: 'lsm9ds0xm' }
	},

	work: function(my) {
		every((1).second(), function() {
			my.accel.getAccel(function(err, data) {
				console.log(data);
			});
		});
	}
}).start();

The errors I got while running it on my edison:
root@FYP_Edison_B:~/programs# node testCylon.js 
I, [2015-06-23T01:53:36.882Z]  INFO -- : [Robot 21276] - Initializing connections.
I, [2015-06-23T01:53:37.377Z]  INFO -- : [Robot 21276] - Initializing devices.
I, [2015-06-23T01:53:37.394Z]  INFO -- : [Robot 21276] - Starting connections.
I, [2015-06-23T01:53:37.402Z]  INFO -- : [Robot 21276] - Starting devices.
I, [2015-06-23T01:53:37.440Z]  INFO -- : [Robot 21276] - Working.

RangeError: Trying to access beyond buffer length
    at checkOffset (buffer.js:582:11)
    at Buffer.readInt16LE (buffer.js:640:5)
    at /home/root/node_modules/cylon-i2c/lib/lsm9ds0xm.js:121:28
    at Adaptor.i2cRead (/home/root/node_modules/cylon-intel-iot/lib/adaptor.js:261:3)
    at LSM9DS0XM.getAccel (/home/root/node_modules/cylon-i2c/lib/lsm9ds0xm.js:112:19)
    at null.<anonymous> (/home/root/programs/testCylon.js:34:13)
    at wrapper [as _onTimeout] (timers.js:261:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)
root@FYP_Edison_B:~/programs# i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- 1d -- -- 
20: UU UU UU UU -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- 6b -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@FYP_Edison_B:~/programs# 

As shown above, the correct addresses for the lsm9ds0xm and lsm9ds0g are detected. I think the problem is with the line "at null.<anonymous> (/home/root/programs/testCylon.js:34:13)", which refers to the line "my.accel.getAccel(function(err, data) {" in my script file testCylon.js. My guess is that it doesn't detect the device at all? Anyone understands what is wrong and how to solve this?

deadprogram, 2015-06-23:
Hi, @fooqri looks to me like the device is showing up on bus 1 for the Sparkfun block. The current cylon-intel-iot adaptor makes some assumptions about the i2c bus number based on which Intel board is being used, that currently only looks on bus 0 or 6, neither of which recognize the Sparkfun blocks. 

However, the change needed to address this is pretty simple, we just need to be able to specify which i2c bus is to be used, and only use the current values as a default in the case none is provided. I think only a 2-line change is required to support this. We'll try to get it handled this week, if possible. Thanks for the detailed issue report, it is appreciated.

deadprogram, 2015-06-23:
Hi, @fooqri 

I just added this branch to the `cylon-intel-iot` repo: https://github.com/hybridgroup/cylon-intel-iot/tree/feature/specify-i2c-port 

It allows you to specify which i2c port to use like this:

```
connections: {
    edison: { adaptor: "intel-iot", i2cPort: 1 }
  },
```

Can you try this and see if it works for you? Thanks!

edgarsilva, 2015-07-09:
@deadprogram looks like that branch has already been merged, should we close?

deadprogram, 2015-07-10:
@fooqri we released an update that should have addressed this. Can you please verify, so we can close this open issue? Thanks!

fooqri, 2015-07-10:
I had to set things up a bit but I have this version installed:
...
 "name": "cylon-i2c",
  "version": "0.22.0",
...

And using this code:
--------------------------
"use strict";

var Cylon = require("cylon");

Cylon.robot({
connections: {
    edison: { adaptor: 'intel-iot', i2cPort: 1 },
  },

  devices: {
    led: { driver: 'led', pin: 13, connection: 'edison' },
    gyro: { driver: "lsm9ds0g" }
  },

  work: function(my) {
    console.log("Setting up Collision Detection...");
    every((1).second(), function() {
        my.gyro.getGyro(function(err, data) {
            console.log(data);
        });
    });
  }
}).start();

I get this error:
-------------------
I, [2015-07-10T00:20:39.725Z]  INFO -- : [Robot 1] - Starting connections.
I, [2015-07-10T00:20:39.743Z]  INFO -- : [Robot 1] - Starting devices.
I, [2015-07-10T00:20:39.756Z]  INFO -- : [Robot 1] - Working.
Setting up Collision Detection...

RangeError: Trying to access beyond buffer length
    at checkOffset (buffer.js:582:11)
    at Buffer.readInt16LE (buffer.js:720:5)
    at /home/root/node_modules/cylon-i2c/lib/lsm9ds0g.js:114:25
    at Adaptor.i2cRead (/home/root/node_modules/cylon-intel-iot/lib/adaptor.js:263:3)
    at LSM9DS0G.getGyro (/home/root/node_modules/cylon-i2c/lib/lsm9ds0g.js:105:19)
    at null.<anonymous> (/home/root/gyro.js:18:17)
    at wrapper [as _onTimeout] (timers.js:258:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)

Hardware:
--------------
Using edison + sparkfun console board + sparkfun i2c board + sparkfun lsm9ds0 board

Is this the correct version of cylon-i2c, it looks like the npm version and the master version are the same version number.

fooqri, 2015-07-10:
Please ignore previous it is indeed working, not sure why it failed on earlier run. All subsequent runs have produced data.

I, [2015-07-10T00:42:08.293Z]  INFO -- : [Robot 1] - Starting connections.
I, [2015-07-10T00:42:08.311Z]  INFO -- : [Robot 1] - Starting devices.
I, [2015-07-10T00:42:08.327Z]  INFO -- : [Robot 1] - Working.
Setting up Collision Detection...
{ x: 30583, y: 30583, z: 30583 }
{ x: 19018, y: 19018, z: 19018 }
{ x: -5912, y: -5912, z: -5912 }
{ x: -30327, y: -30327, z: -30327 }

fooqri, 2015-07-10:
Please see my updated comment above, I believe one more change is needed for it to work. Adaptor.prototype.i2cWrite also need to use the i2cPort value.

edgarsilva, 2015-07-10:
@fooqri seems like you are right gonna check it out tomorrow morning to replicate and fix the issue based on your feedback, thanks.

fooqri, 2015-08-04:
The data from the accelerometer seems strange, as the x, y, z values are always the same, and they change fairly randomly when device is at rest. 

{ x: 30583, y: 30583, z: 30583 }
{ x: 19018, y: 19018, z: 19018 }
{ x: -5912, y: -5912, z: -5912 }
{ x: -30327, y: -30327, z: -30327 }

I am wondering if perhaps something isn't being properly initialized in _LSM9DS0XM.prototype._initAccel()_ in __lsm9ds0xm.js__. I will try to take a look but if you have any advice it would be greatly appreciated. Is this the spec you used for the init and read https://www.nordevx.com/content/lsm9ds0-accelerometer-magnetometer-and-gyro-one for the settings?

deadprogram, 2015-08-04:
@edgarsilva do you have this hardware available for testing?

edgarsilva, 2015-08-05:
@deadprogram I do not have that accelerometer. I have an Adafruit LSM303 and a ADXL345, anyone knows if one of these two might help me debug this?