hybridgroup/cylon-i2c #37: Device agnostic driver

chintanp, 2015-12-02:
Is there no way to use this module for a device not part of the driver list ?

I have a TI fuel gauge (BQ-27441), which is a battery sensor that reads battery information like current, voltage , temp etc. of the battery. I have connected it to my ras-pi, via I2C. 

Any way I can leverage this module for use with that ?

deadprogram, 2015-12-03:
Seems like this would not be difficult, in similar fashion to the `directPin` driver in `cylon-gpio`. We will look into doing this.

deadprogram, 2016-04-25:
Hi @chintanp it took longer than expected, but please check out the new DirectI2C driver https://github.com/hybridgroup/cylon-i2c/blob/master/lib/direct-i2c.js

chintanp, 2016-04-25:
@deadprogram : Thank you for this. I am away from my system, so I cant test this, but a sample program would be very helpful. 

Thanks.

closed 2016-04-25 by chintanp