hybridgroup/cylon-i2c #41: Is my screen incompatible ?

georges6410, 2016-03-16:
Hello, 

I've bought a kit with some components, and in this kit there was a 1602 LCD screen (16*2) : [kit here](http://www.amazon.fr/Wallzkey-%C3%89lectronique-Package-Composant-Arduino/dp/B010G3ZZBO/ref=sr_1_4?ie=UTF8&qid=1458165336&sr=8-4&keywords=kit+electronique)

After trying this in my code : 

`lcdScreen: {driver: 'lcd'}`

and this : 

`my.lcdScreen.print("test");`

It doesn't work at all.

So I've tried using this [LCD Library](https://www.npmjs.com/package/lcd), and with this code it worked.

So I just wanted to know if I'm doing something bad, or if it's not compatible at all.

Thanks

deadprogram, 2016-03-16:
Hi, @georges6410 as of right now, the cylon-i2c module only supports i2c devices. Your screen uses direct GPIO communication, as you already suspected.

georges6410, 2016-03-18:
Okay, thanks for the information, so I have to find a screen compatible with i2c drivers.
Any hints on low-budget lcd screens ?

mvkarthikeyan, 2016-09-22:
man can i display OLED (128*64 monochrome) using I2C