hybridgroup/cylon-i2c #42: BMP180 not working

jvelazqueztraut, 2016-04-16:
I couldn't make the BMP180 example run. I'm using the BMP version with only 4 pin and an RPi3, though I'm guessing those two should not make a difference.
I was using the example code found in cylon website, and after compiling there were no measurements at all. Temp at 0 and pressure at infinite. Later, I added the bmp180.start before taking measurements and I started seeing some rawTemp numbers, but they just didn't make sense.

deadprogram, 2016-04-18:
Hi @jvelazqueztraut our BMP180 is checked out of the workshop, but we'll look into it.

What hardware platform are you using?

jvelazqueztraut, 2016-04-18:
Raspberry Pi (B+ and 3, tried it on both). The i2c communications seems to be working fine since the device is present on i2cdetect.

deadprogram, 2016-04-18:
You had already mentioned Raspi3 in your original issue, sorry I overlooked that. In any case, we'll try to repro when the BMP180 is back later today.

deadprogram, 2016-04-20:
Hi @jvelazqueztraut I've just pushed some fixes for the BMP180 driver to the master branch that should address your issue. Please give it a try.

```

```

deadprogram, 2016-04-25:
Hi @jvelazqueztraut did you ever have a chance to take a look at the updates? Please let us know.

jvelazqueztraut, 2016-04-26:
Hi @deadprogram, just tried this. Parsing seems to be wrong. I attach my log.
[image]

deadprogram, 2016-04-26:
When I ran the program https://github.com/hybridgroup/cylon-i2c/blob/master/examples/bmp180/bmp180.js just now on my Edison, I received this output:

```
root@howotocode:~/cylon# node bmp180.js                                                                                
2016-04-26T18:39:53.782Z : [Robot 1] - Starting connections.
2016-04-26T18:39:53.862Z : [Robot 1] - Starting connection 'board'.
2016-04-26T18:39:53.865Z : [Robot 1] - Starting devices.
2016-04-26T18:39:53.866Z : [Robot 1] - Starting device 'bmp180'.
2016-04-26T18:39:53.903Z : [Robot 1] - Working.
{ temp: 114.3 }
{ temp: 22.7, press: 99867, alt: 122.10182064441162 }
{ temp: 22.7 }
{ temp: 22.7, press: 99861, alt: 122.60727095277399 }
{ temp: 22.7 }
{ temp: 22.7, press: 99855, alt: 123.1127458517038 }
{ temp: 22.7 }
{ temp: 22.7, press: 99861, alt: 122.60727095277399 }
{ temp: 22.7 }
{ temp: 22.7, press: 99858, alt: 122.86000532825001 }
{ temp: 22.7 }
{ temp: 22.7, press: 99849, alt: 123.61824534386861 }
{ temp: 22.7 }
{ temp: 22.7, press: 99861, alt: 122.60727095277399 }
{ temp: 22.7 }
{ temp: 22.7, press: 99861, alt: 122.60727095277399 }
...
```

This is basically the same output I got on my C.H.I.P. so seems like driver is ok? Can you please share the source code you are using, or at least the parts calling the bmp180?

jvelazqueztraut, 2016-04-26:
I'm using this code copied from https://cylonjs.com/documentation/drivers/bmp180/

`var Cylon = require('cylon');

Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    bmp180: { driver: 'bmp180' }
  },

  work: function(my) {
    my.bmp180.getTemperature(function(err, val) {
      if (err) {
        console.log(err);
        return;
      }

      console.log("getTemperature call:");
      console.log("\tTemp: " + val.temp + " C");
    });

    after((1).second(), function() {
      my.bmp180.getPressure(1, function(err, val) {
        if (err) {
          console.log(err);
          return;
        }

        console.log("getPressure call:");
        console.log("\tTemperature: " + val.temp + " C");
        console.log("\tPressure: " + val.press + " Pa");
      });
    });

    after((2).seconds(), function() {
      my.bmp180.getAltitude(1, null, function(err, val) {
        if (err) {
          console.log(err);
          return;
        }

        console.log("getAltitude call:");
        console.log("\tTemperature: " + val.temp + " C");
        console.log("\tPressure: " + val.press + " Pa");
        console.log("\tAltitude: " + val.alt + " m");
      });
    });
  }
}).start();`

deadprogram, 2016-04-26:
That code is out of date with the repo, please run the example from the link I provided.

jvelazqueztraut, 2016-04-26:
[image]

deadprogram, 2016-04-27:
Hi, @jvelazqueztraut I finally thought "hmm, I wonder if this is actually a problem with the cylon-raspi adaptor?" So I plugged in my Raspi3 and turns out that was indeed the problem. 

```
pi@raspberrypi:~/Development/robots $ node bpm180.js 
2016-04-27T01:04:16.027Z : [Robot 1] - Starting connections.
2016-04-27T01:04:16.075Z : [Robot 1] - Starting connection 'board'.
2016-04-27T01:04:16.083Z : [Robot 1] - Starting devices.
2016-04-27T01:04:16.084Z : [Robot 1] - Starting device 'bmp180'.
2016-04-27T01:04:16.096Z : [Robot 1] - Working.
{ temp: 23.6 }
{ temp: 23.6, press: 99739, alt: 132.89009771694398 }
{ temp: 23.6, press: 99739 }
{ temp: 23.6 }
{ temp: 23.6, press: 99726, alt: 133.9864090924152 }
{ temp: 23.6, press: 99726 }
{ temp: 23.6 }
{ temp: 23.6, press: 99726, alt: 133.9864090924152 }
{ temp: 23.6, press: 99726 }
{ temp: 23.6 }
{ temp: 23.6, press: 99726, alt: 133.9864090924152 }
{ temp: 23.6, press: 99726 }
...
```

I just released cylon-raspi v0.20.1 please give that a try.

deadprogram, 2016-05-09:
Hi @jvelazqueztraut if you have further troubles, please reopen this issue. Otherwise, I think it is fixed in the last release. Thanks again for your help.

closed 2016-05-09 by deadprogram